atomic14/diy-alexa #2: build error

AIchar, 2020-10-27:
hi，
I got a compilation error and he showed

lib\tfmicro/tensorflow/lite/kernels/internal/max.h:29:10: error: 'fmax' is not a member of 'std'
lib\tfmicro/tensorflow/lite/kernels/internal/min.h:29:10: error: 'fmin' is not a member of 'std'

cgreening, 2020-10-27:
Hi There,

I just checked on a fresh checkout from git and can't recreate. Can you tell me which version of platform.io you are using?

I'm using: PlatformIO, version 5.0.1

Thanks
Chris

awal123, 2020-10-27:
Hi,

I had the same problem and found this solution (works for me): https://community.platformio.org/t/libraries-not-detected-cannot-open-source-file/16054/4

Best regards,
Artur

AIchar, 2020-10-27:
Hi Chris,

The operating system I use is win10.
I'm using: PlatformIO IDE for VSCode, version 2.1.3
It seems to be the latest
[image]
[image]

Thanks
Arzhe

cgreening, 2020-10-27:
Hi Arzhe,

I've tried adding a flag to the library.json to make the code use `std::max` instead of `std::fmax`. I'm not sure why this would work on my machine (a mac) and not windows.

Let me know if this fixes it.

Thanks
Chris

AIchar, 2020-10-27:
> Hi Arzhe,
> 
> I've tried adding a flag to the library.json to make the code use `std::max` instead of `std::fmax`. I'm not sure why this would work on my machine (a mac) and not windows.
> 
> Can you pull the latest code and let me know if this fixes it.
> 
> Thanks
> Chris

The latest code still gets compilation errors.
But I added defined(arduino) to max.h and min.h to solve this problem.I can wake up esp32 through "Marven" now

AIchar, 2020-10-28:
> Hi,
> 
> I had the same problem and found this solution (works for me): https://community.platformio.org/t/libraries-not-detected-cannot-open-source-file/16054/4
> 
> Best regards,
> Artur

Hi Artur,

It work for me too. I can compile it now

Thank

closed 2020-10-29 by AIchar