atomic14/diy-alexa #3: Different I2S Connection at M5Stack Atom Echo...any chance to adapt this?

sthufnagl, 2020-11-01:
Hi,

I currently try to port your "diy-alexa" to an Atom Echon from M5Stack.  Atom Echo because it it very small.
The I2S connection diagram of the Atom is simpler and I wonder if it fits to your diy-alexa. (I2S_MIC_LEFT_RIGHT_CLOCK is missing!)
[image]

What I changed was the pin mapping at config.h like below...but it doesn't work.
```
// Which channel is the I2S microphone on? I2S_CHANNEL_FMT_ONLY_LEFT or I2S_CHANNEL_FMT_ONLY_RIGHT
#define I2S_MIC_CHANNEL I2S_CHANNEL_FMT_ONLY_LEFT

#define I2S_MIC_SERIAL_CLOCK GPIO_NUM_33
#define I2S_MIC_SERIAL_DATA GPIO_NUM_23

// Analog Microphone Settings - ADC1_CHANNEL_7 is GPIO35
#define ADC_MIC_CHANNEL ADC1_CHANNEL_7

// speaker settings
#define I2S_SPEAKER_SERIAL_CLOCK GPIO_NUM_19
#define I2S_SPEAKER_LEFT_RIGHT_CLOCK GPIO_NUM_33
#define I2S_SPEAKER_SERIAL_DATA GPIO_NUM_22
...
```

Any Chance to get "diy-alexa" running with the Atom Echo? 

Thanks in advance

Steve

cgreening, 2020-11-01:
Hey Steve,

I think they are sharing the same I2S interface for both the speaker and the microphone. I'm guessing that they must have the BCLK connected internally as I'm not sure how it can work otherwise.

Looking at the sample code here. It looks like you can't use the speaker at the same time as the mic. There's a couple of issues, it looks like both the speaker and the mic are on the right channel - so they will conflict, and the mic is using PDM format which won't work with the speaker.

https://github.com/m5stack/M5-ProductExampleCodes/blob/master/Core/Atom/AtomEcho/Arduino/Repeater/Repeater.ino

This shouldn't be a problem as we don't really need to listen when playing sound.

It will need a bit of a fundamental change in how the code currently works though as the code currently uses the samples coming from the microphone to drive the state machine even when not listening.

Let me have a think if the code can be easily refactored to do this as it looks like it should be possible.

sthufnagl, 2020-11-01:
Hi,
could it be that the "Mic CLK" is the I2S_MIC_LEFT_RIGHT_CLOCK? If yes then the I2S_MIC_SERIAL_CLOCK is missing or unused?
Thx

cgreening, 2020-11-01:
Yes, mic clk is the I2S_MIC_LEFT_RIGHT_CLOCK. Here's the pins from their sample code:

```
#define CONFIG_I2S_BCK_PIN 19
#define CONFIG_I2S_LRCK_PIN 33
#define CONFIG_I2S_DATA_PIN 22
#define CONFIG_I2S_DATA_IN_PIN 23
```

I think you are correct - the mic does not need a serial clock as it is using PDM output which I think doesn't need a serial clock.

I will see if I can get hold of atom echo and see if I can get it to work.

We'll need to work out how to change the code so that the I2S input stops to do the output. This might be a bit fiddly but could just fit in with how the current state machine works.

sthufnagl, 2020-11-01:
Hi,
the Atom Echo is very, very cheap but don't expect much from the internal speaker. I will try to print a small "Gramaphone Horn" to amplify the Atom Speaker Volume.
BTW, the hint with PDM and "switching" Mic and Speaker is very valueable. I will follow this hint.

Thx

cgreening, 2020-11-01:
ok, will take around 14 days for it to arrive, meanwhile I'll have a think about how to reorganise the code.

sthufnagl, 2020-11-01:
Hi,
wouldn't we save GPIO Pins if we use the same "Atom Echo I2S Wiring"?
I ask because I want to use a JTAG Debugger and currently the GPIO12 and 14 are occupied .
Thx and lets continue in 2 weeks...

cgreening, 2020-11-02:
Hey Steve,

You should be able to move the pins around - I think the I2S interface can work on any pin.

The only issue with the Atom Echo wiring is that you can't use the microphone and speaker at the same time - so you need to switch between the two - probably not a big issue, but annoying with the code as it :)

Cheers
Chris

sthufnagl, 2020-11-03:
Hi Chris,

I used the other code from your "voice-controlled robot" repo because it was easier to adapt to Atom Echo Mic.
It worked!
I commited the changes to my fork of your repo. 
:-)
Next will be the Atom Echo Speaker...

Cheers
Steve

sthufnagl, 2020-11-24:
Solved...thanks for the support.
:-)

closed 2020-11-24 by sthufnagl