atomic14/diy-alexa #5: Directional microphone

StuartIanNaylor, 2020-11-10:
Again not an issue but just wondering how much load an ESP32 can take but I2S is 2 channel so was wondering if digitally you could do something similar to this.

https://invensense.tdk.com/wp-content/uploads/2015/02/Low-Noise-Directional-Studio-Microphone-Reference-Design1.pdf

_Beamforming
Beamforming involves processing the output of multiple microphones (or in this case, multiple mic arrays) to create a directional
pickup pattern. For recording and live sound applications it is important that the microphone only picks up sound from one
direction, such as from the singer or instrument, and attenuates the sound that is off the main axis. Beamforming is implemented in
this design using analog delays, an equalization filter, and a summing amplifier.
A two-element array is set up by placing two microphone boards distance, d, apart. A cardioid pattern
(Figure 4) is achieved by delaying the signal from one array board by amount of time it takes sound to travel between the two
boards, and subtracting this delayed signal from the signal from the first microphone array board. With this type of spatial response,
the microphone rejects sounds from the sides and rear, while picking up sounds incident to the front of the microphone._

We don't have the mic clusters but the stereo pair could be the two-element array with the delay of mic distance and subtraction done digitally?
Its just the delay part of https://hackaday.io/project/162628-audio-delay-and-vox-using-esp32 minus the vox.

Does the ESP32 lack the memory allocation to provide a stereo I2S input of the initial short delay of the mic distance of the speed of sound then enter the KW ring buffer?

Its poor mans beamforming but for many the improvement of having a directional mic rather than omnidirectional pickup of all is a big plus for far field.

cgreening, 2020-11-11:
I think you could definitely do it at sample capture time - as you say - before putting the samples into the ring buffer.

If memory does become an issue then I'd be tempted to look at the wrover modules which have 4MB of RAM.

StuartIanNaylor, 2020-11-11:
Yeah my wondering mind keeps thinking that a distributed array of directional mics that use the KW hit result could give a central ASR a signal to use the best and nearest mic signal.
But yeah the wrover dont have much price difference now.
https://www.ebay.co.uk/itm/ESP32-WROVER-B-T8-V1-8-ESP32-8MB-PSRAM-TF-Card-WiFi-Module-Bluetooth-Board-HL/293784167596
Would need 2x i2s mics but still very cheap.

Does anyone know if there is a lib for a lossless audio RTP with latency compensation similar to snapcast?

StuartIanNaylor, 2020-11-26:
Whilst we are talking mics has anyone tried the MASE and other algs on Wrover or lower?
https://github.com/espressif/esp-sr

If it was a RTP mic streamer triggered by KW till silence that would be amazing to couple to a central ASR.
If it maybe streamed AMR-WB to a port and accompanied KWS hit score on another then multiple ESP32 could be part of a distributed array.

closed 2020-11-26 by StuartIanNaylor

andrej102, 2021-02-17:
I tested MASE. It's just a noise reduction algorithm, it has nothing to do with beamforming. At the same time, it does not work well - it greatly distorts the sound and cuts the frequencies to 4 kHz. In the future, the automatic voice recognition system recognizes the voice much worse than without MUSE.

StuartIanNaylor, 2021-02-17:
@andrej102 I did wonder how they had shoehorned a series of load heavy DSP that even seems to be too much for Raspberry embedded Linux. 4kHz is no use to the central ASR I plan to use.

They have done an amazing demo with the esp32-s3 https://www.youtube.com/watch?v=ARkuaeEW1eM

The EQ DSP to rectify the filtering of the beamforming also adds much load that the stereo mic idea I gave up on.
The ADC on the ESP32 from bits to SNR is not great but going analogue with directional electrets is an alternative.

The above vid has me confused as it seems to exceed high end dsp beamformers I have seen and if snakeoil or not I am unsure.
I still plan to use distributed directional electrets so that a unit is voice=near / noise=far purely by positioning