atomic14/diy-alexa #8: Hi Chris

StuartIanNaylor, 2020-12-14:
I got an esp32 audio kit as its just a wrover with a codec built in https://www.banggood.com/ESP32-Aduio-Kit-WiFi-bluetooth-Module-ESP32-Serial-to-WiFi-Audio-Development-Board-with-ESP32-A1S-p-1449256.html

£13 not too pricey...

I got the ADF working which they just call https://github.com/Ai-Thinker-Open/ESP32-A1S-AudioKit its just download the toolchain set adf path to this and the idf path to the idf contained.

I have run a few of the examples and the complete ADF seems to work even if the onboard mics seem to be extremely insensitive.

I just wondered if you had done the same and tinkered with the ADF and maybe grasped how to set input volume or the ALC as seem to have it running but damned if I can tell the difference :)

Have you given them a go and the ADF as from about £9 they are at an interesting price point.

cgreening, 2020-12-14:
I've ordered one to have a play.

StuartIanNaylor, 2020-12-14:
Yeah apols from me if it turns out the gain is just pretty poor but not the end of the world as its just a matter of using the line in.
I have been using a Max9814 and yeah they are a bit noisy with the float level gain selected or have used these before but can be a tad quiet on a line in.

https://www.ebay.co.uk/itm/ADMP401-MEMS-Microphone-Breakout-Module-Board-For-Arduino-Universal-1-3cm-1cm-UK/192101467145 they are analogue but great SNR.

There is a lot to go at with the ADF with really only a brief API reference and found the quite numerous examples essential as would be stumped otherwise.
PS don't do an Alexa just do KWS with broadcast :)

closed 2020-12-14 by StuartIanNaylor

StuartIanNaylor, 2020-12-14:
PS I failed trying on windows and on Ubuntu 18.04 with python2 & the https://docs.espressif.com/projects/esp-idf/en/v3.3.1/get-started/linux-setup.html toolchain went without a hitch.

StuartIanNaylor, 2020-12-16:
PS Chris give these another try as seriously they are excellent these ones have jumper and there own ldo.

https://www.ebay.co.uk/itm/253051508078

Tested and apart from the top gain with good SNR on the pi with a soundcard (top gain float slightly hissy though)