atomic14/diy-alexa #9: Running into issues

MikoJK, 2020-12-22:
Heya.

I tried to test deploy the repo to my ESP32 (Lolin D32) but I ran into several issues:

- There was no .ino file so I renamed main.cpp to src.ino as Visual Studio Code (with Arduino plugin) and Arduino IDE both complained lack of .ino file, is this ok?
- There were a lot of include errors and despite adding includePaths to c_cpp_propertier.json, it still didn't work so I added the header files onto src-folder, is this ok as well?
- After doing so, I no longer got include errors but instead of this:

`src:100:33: error: expected type-specifier before 'I2SMicSampler'

   I2SSampler *i2s_sampler = new I2SMicSampler(i2s_mic_pins, false);`

I can't seem to figure out what is the cause of this. I'm pretty sure I wasn't supposed to change those file names and locations but I couldn't get rid of the errors in any other means. Any tips?

cgreening, 2020-12-22:
Hi Miko,

Apologies for not making it clearer.

You need to use the platformio - https://platformio.org/ plugin to build the code. It's not designed for use with the Arduino IDE or plugin.

That should fix your issues.

Best
Chris

MikoJK, 2020-12-22:
Ohh I see. 

After installing that and uninstalling the Arduino plugin, It worked. Thank you so much!

closed 2020-12-22 by MikoJK