atomic14/diy-alexa #11: Circuit Sketch Diagram Required

arijitaich, 2021-03-01:
Greetings,

My project is somehow not working properly and i think its because of the circuit setup i may have done incorrectly.
Could you please provide a circuit sketch so that the connections become clear.

Thanks.

cgreening, 2021-03-02:
Hi There,

What components are you using - microphone board and speaker amplifier?

Thanks

pureskar, 2021-03-02:
**Thanks for the response. Here are the list of components:**

1 x  INMP441 MEMS Omnidirectional Microphone Module High Precision/SNR Low Power I2C Interface Supports ESP32

1 x  ESP32 Development Board WiFi+Bluetooth

1 x  I2S Audio Breakout - MAX98357A (Sparkfun USA)

arijitaich, 2021-03-02:
**Thanks for the response. Here are the list of components:**

1 x INMP441 MEMS Omnidirectional Microphone Module High Precision/SNR Low Power I2C Interface Supports ESP32

1 x ESP32 Development Board WiFi+Bluetooth

1 x I2S Audio Breakout - MAX98357A (Sparkfun USA)

BTW both arijitaich & puresker are my accounts.

closed 2021-03-18 by arijitaich