atomic14/diy-alexa #12: Outputs Average detection time 95ms. But system isn't working.

arijitaich, 2021-03-08:
I have tried uploading the sketch but on successful upload, This is what happens:

--- Quit: Ctrl+C | Menu: Ctrl+T | Help: Ctrl+T followed by Ctrl+H ---
ts Jun  Starting up
Total heap: 312308
Free heap: 236128
E (2513) SPIFFS: mount failed, -10025
[E][SPIFFS.cpp:89] begin(): Mounting SPIFFS failed! Error: -1
[E][vfs_api.cpp:22] open(): File system is not mounted
ERROR: bit depth 16379 is not supported please use 16 bit signed integer
ERROR: bit depth 200 is not supported please us 16KHz
fmt_chunk_size=0, audio_format=0, num_channels=0, sample_rate=200, sample_alignment=7984, bit_depth=16379, data_bytes=1073462756
[E][vfs_api.cpp:22] open(): File system is not mounted
ERROR: bit depth 200 is not supported please us 16KHz
fmt_chunk_size=0, audio_format=0, num_channels=0, sample_rate=200, sample_alignment=7984, bit_depth=16379, data_bytes=1073462756
[E][vfs_api.cpp:22] open(): File system is not mounted
ERROR: bit depth 200 is not supported please us 16KHz
fmt_chunk_size=0, audio_format=0, num_channels=0, sample_rate=200, sample_alignment=7984, bit_depth=16379, data_bytes=1073462756
[E][vfs_api.cpp:22] open(): File system is not mounted
ERROR: bit depth 200 is not supported please us 16KHz
fmt_chunk_size=0, audio_format=0, num_channels=0, sample_rate=200, sample_alignment=7984, bit_depth=16379, data_bytes=1073462756
[E][vfs_api.cpp:22] open(): File system is not mounted
ERROR: bit depth 200 is not supported please us 16KHz
fmt_chunk_size=0, audio_format=0, num_channels=0, sample_rate=200, sample_alignment=7984, bit_depth=16379, data_bytes=1073462756
[E][vfs_api.cpp:22] open(): File system is not mounted
ERROR: bit depth 200 is not supported please us 16KHz
fmt_chunk_size=0, audio_format=0, num_channels=0, sample_rate=200, sample_alignment=7984, bit_depth=16379, data_bytes=1073462756
[E][vfs_api.cpp:22] open(): File system is not mounted
ERROR: bit depth 200 is not supported please us 16KHz
fmt_chunk_size=0, audio_format=0, num_channels=0, sample_rate=200, sample_alignment=7984, bit_depth=16379, data_bytes=1073462756
[E][vfs_api.cpp:22] open(): File system is not mounted
ERROR: bit depth 200 is not supported please us 16KHz
fmt_chunk_size=0, audio_format=0, num_channels=0, sample_rate=200, sample_alignment=7984, bit_depth=16379, data_bytes=1073462756
[E][vfs_api.cpp:22] open(): File system is not mounted
ERROR: bit depth 200 is not supported please us 16KHz
fmt_chunk_size=0, audio_format=0, num_channels=0, sample_rate=200, sample_alignment=7984, bit_depth=16379, data_bytes=1073462756
Loading model
12 bytes lost due to alignment. To avoid this loss, please make sure the tensor_arena is 16 bytes aligned.
Used bytes 22604

Created Neral Net
m_pooled_energy_size=43
Created audio processor
Starting i2s
Average detection time 95ms
Average detection time 95ms **- REPEATS.**

Here is my hardware:
1 x INMP441 MEMS Omnidirectional Microphone Module High Precision/SNR Low Power I2C Interface Supports ESP32

1 x ESP32 Development Board WiFi+Bluetooth

1 x I2S Audio Breakout - MAX98357A (Sparkfun USA)

cgreening, 2021-03-08:
There's a couple of errors that indicate that SPIFFS has not been setup:

```
E (2513) SPIFFS: mount failed, -10025
[E][SPIFFS.cpp:89] begin(): Mounting SPIFFS failed! Error: -1
```

Have you uploaded the filesystem? That contains all the WAV files. Once we've fixed those errors we can have a look at why it's not doing anything.

arijitaich, 2021-03-08:
You mean the 3 wav files? turn_off.wav, turn_on.wav, & turn_on2.wav ?

arijitaich, 2021-03-08:
Uploaded the filesystem as well. Now here is the output:

--- Miniterm on COM4  115200,8,N,1 ---
--- Quit: Ctrl+C | Menu: Ctrl+T | Help: Ctrl+T followed by Ctrl+H ---
s JuStarting up
Total heap: 312324
Free heap: 236144
fmt_chunk_size=16, audio_format=1, num_channels=1, sample_rate=16000, sample_alignment=2, bit_depth=16, data_bytes=24892 
fmt_chunk_size=16, audio_format=1, num_channels=1, sample_rate=16000, sample_alignment=2, bit_depth=16, data_bytes=32000 
fmt_chunk_size=16, audio_format=1, num_channels=1, sample_rate=16000, sample_alignment=2, bit_depth=16, data_bytes=12226 
fmt_chunk_size=16, audio_format=1, num_channels=1, sample_rate=16000, sample_alignment=2, bit_depth=16, data_bytes=169042
fmt_chunk_size=16, audio_format=1, num_channels=1, sample_rate=16000, sample_alignment=2, bit_depth=16, data_bytes=157264
fmt_chunk_size=16, audio_format=1, num_channels=1, sample_rate=16000, sample_alignment=2, bit_depth=16, data_bytes=147434
fmt_chunk_size=16, audio_format=1, num_channels=1, sample_rate=16000, sample_alignment=2, bit_depth=16, data_bytes=160072
fmt_chunk_size=16, audio_format=1, num_channels=1, sample_rate=16000, sample_alignment=2, bit_depth=16, data_bytes=147436
fmt_chunk_size=16, audio_format=1, num_channels=1, sample_rate=16000, sample_alignment=2, bit_depth=16, data_bytes=164284
Loading model
8 bytes lost due to alignment. To avoid this loss, please make sure the tensor_arena is 16 bytes aligned.
Used bytes 22592

Created Neral Net
m_pooled_energy_size=43
Created audio processor
Starting i2s
Average detection time 98ms
Average detection time 98ms
Average detection time 98ms
Average detection time 99ms
Average detection time 98ms
Average detection time 98ms
Average detection time 98ms **REPEATS**

cgreening, 2021-03-08:
ok, so everything is working, but I think the audio is not getting from the microphone into the system so the wakeword is not triggering - I'm assuming that you are saying "Marvin" to it.

Can you check the wiring of your microphone against the details in `config.h`? and make sure it's correctly wired up?

arijitaich, 2021-03-08:
Since this is the first time I am doing it, I have kept everything similar to the process you have followed. Didn't want to miss out on anything.

Yes, I took the trigger word as Marvin.

Here are my current connections:

ESPWROOM32                     INMP441
GND -------------------------->GND
VDD---------------------------->3V3
D5------------------------------>SCK
D18----------------------------->SD
D19----------------------------->WS

ESPWROOM32                        I2S Audio Breakout - MAX98357A
D12----------------------------------->LRCLK
D14----------------------------------->BCLK
D27----------------------------------->DIN

arijitaich, 2021-03-08:
**Here is my config.h. I think i have set it correctly.**

// WiFi credentials
#define WIFI_SSID "OTTO"
#define WIFI_PSWD "sadasd"

// are you using an I2S microphone - comment this out if you want to use an analog mic and ADC input
#define USE_I2S_MIC_INPUT

// I2S Microphone Settings

// Which channel is the I2S microphone on? I2S_CHANNEL_FMT_ONLY_LEFT or I2S_CHANNEL_FMT_ONLY_RIGHT
#define I2S_MIC_CHANNEL I2S_CHANNEL_FMT_ONLY_LEFT
// #define I2S_MIC_CHANNEL I2S_CHANNEL_FMT_ONLY_RIGHT
#define I2S_MIC_SERIAL_CLOCK GPIO_NUM_33
#define I2S_MIC_LEFT_RIGHT_CLOCK GPIO_NUM_26
#define I2S_MIC_SERIAL_DATA GPIO_NUM_25

// Analog Microphone Settings - ADC1_CHANNEL_7 is GPIO35
#define ADC_MIC_CHANNEL ADC1_CHANNEL_7

// speaker settings
#define I2S_SPEAKER_SERIAL_CLOCK GPIO_NUM_14
#define I2S_SPEAKER_LEFT_RIGHT_CLOCK GPIO_NUM_12
#define I2S_SPEAKER_SERIAL_DATA GPIO_NUM_27

// command recognition settings
#define COMMAND_RECOGNITION_ACCESS_KEY "LJUX3BNHM22LJVDG4HA32LG6DBQCPS2E"

cgreening, 2021-03-08:
I think there's a mismatch between your wiring and what's in the config.

ESPWROOM32 INMP441
GND -------------------------->GND
VDD---------------------------->3V3
D5------------------------------>SCK
D18----------------------------->SD
D19----------------------------->WS

You should change the config.h file to match your wiring (or change the wiring to match the config.h file - probably easier to change the config?)

Update the config to:

#define I2S_MIC_CHANNEL I2S_CHANNEL_FMT_ONLY_LEFT
#define I2S_MIC_SERIAL_CLOCK GPIO_NUM_5
#define I2S_MIC_LEFT_RIGHT_CLOCK GPIO_NUM_19
#define I2S_MIC_SERIAL_DATA GPIO_NUM_18

That should match your wiring.

arijitaich, 2021-03-08:
Tried. Nothing Works. Its the same. :(
Is there a way to test out the mic?
I tried taking Analog readings but I2S.h isnt loading for some reason, but drivers/i2s.h gets loaded but throws some error later related to I2S not being declared.
Is there any simpler way to test em out?

cgreening, 2021-03-08:
You can use the sample code here to test your microphone:

https://github.com/atomic14/esp32_audio

You can also add some `Serial.printf` at various points in the code to see if data is coming through from the microphone. Can you post a photo of the connections from the microphone to the dev board and your `config.h`?

arijitaich, 2021-03-08:
Here as you requested:
https://osmo.tech/core/workspace/MachDelivery/20210308_192837.jpg
https://osmo.tech/core/workspace/MachDelivery/20210308_192904.jpg
https://osmo.tech/core/workspace/MachDelivery/20210308_192913.jpg

Attaching a video for better understanding the wiring:
https://osmo.tech/core/workspace/MachDelivery/esp32.mp4

cgreening, 2021-03-08:
I think from the video you have the RED wire going to 3v3 and the GREY wire going to GND on the ESP32.

It's a bit hard to tell from the video. But I think you have RED and BLACK going to VDD and GND on the microphone - that doesn't look quite right unless the wires are getting swapped somewhere else.

arijitaich, 2021-03-11:
Here is the video.
https://drive.google.com/file/d/1KPbfZSdJNCyTvEh9WEMxjBeRBNXL7GlJ/view?usp=sharing

 I have eliminated all extra connections except the Esp32 connecting with INMP441 .
I really think this to be a connection issue. It would be really really really helpful if you could please provide a Circuit Sketch Diagram. This will eliminate the chances of error drastically not only for me but also for other enthusiasts who want their own AI.

cgreening, 2021-03-11:
Hi AI,

The problem is everyone is using different microphone boards and different pins, but I will try and put something together.

Depending on your version of the INMP441 you may need to tie the L/R pin to Ground so the samples come out on the left channel (which is what the code expects).

The wiring should be:

|INMP441|ESP32|
|---|---|
|Vcc|3v3|
|Gnd|Gnd|
|L/R|Gnd|
|WS|I2S_MIC_LEFT_RIGHT_CLOCK|
|SCK|I2S_MIC_SERIAL_CLOCK|
|SD|I2S_MIC_SERIAL_DATA|

I'm testing it right now and there seem to be some issues with connecting to Wit.AI, but the wake word detection seems to be working fine. It's probably the L/R pin that is causing you problems. There are a lot of microphone boards to keep track of.

cgreening, 2021-03-11:
Wit.ai is now fixed - just pushed up a fix

arijitaich, 2021-03-11:
Ok. So i have figured out the pins for Esp WRoom 32

I2S_MIC_SERIAL_CLOCK  is D22
I2S_MIC_SERIAL_DATA  is D21

But i can't find the I2S_MIC_SERIAL_DATA. There are only 2 I2C channels in the board.
Here is a reference i am using: https://microcontrollerslab.com/wp-content/uploads/2019/02/ESP32-pinout.png

cgreening, 2021-03-11:
Don't forget that on an ESP32 you can map pretty much any PIN to what you want - so you don't need to use the pins in the pinout (there are some exceptions to this but for I2S you are free).

So you can pick a pin to use the I2S_MIC_LEFT_RIGHT_CLOCK - you just need to make sure that you have wired it to the WS pin on the INMP441.

This is true of the other PINS on the INMP441 - you just need to connect them to a pin on the ESP32 and then tell the ESP32 which pin you have connected. That's what the config.h file is for in my code - to tell the ESP32 which pins you have connected the microphone to.

This is the critical part - you can connect the microphone pins anywhere - you just need to update the `config.h` to match the pins you have connected it to.

I think your problem will turn out to be the L/R pin - connect that to GND as well and it should start working.

arijitaich, 2021-03-11:
Ok. I followed those instructions but in vain. Actually i wanted to try out on the ICS-43434 I2S MEMS Microphone Breakout Board
 but this particular is not available here. If i ship it from outside, the cost is going drastically high.
 
 You say you have tested it out for INMP441. Thanks a ton for going out of your busy schedule and doing that. 
I think I am at the end of the entire thing and this is the only step that's creating a barrier.

Once i complete this successfully, i would try to regain the same results in a custom word. Trying to make it a Sanskrit word since the ML follows image pattern detection of spectograms, i think i would be able to generate good results provided i can feed it with a good amount of training  data.

Now this is my hardware connections:

INMP441 --------------------------- ESPWROOM32
SD ---------------------------------- D21
VCC --------------------------------- 3V3
GND -------------------------------- GND
SCK --------------------------------- D22
WS ---------------------------------- D19
L/R ---------------------------------- GND

Here is my config:
// WiFi credentials
#define WIFI_SSID "OTTO_2.4G"
#define WIFI_PSWD "ABC1234"

// are you using an I2S microphone - comment this out if you want to use an analog mic and ADC input
#define USE_I2S_MIC_INPUT

// I2S Microphone Settings

// Which channel is the I2S microphone on? I2S_CHANNEL_FMT_ONLY_LEFT or I2S_CHANNEL_FMT_ONLY_RIGHT
#define I2S_MIC_CHANNEL I2S_CHANNEL_FMT_ONLY_LEFT
// #define I2S_MIC_CHANNEL I2S_CHANNEL_FMT_ONLY_RIGHT
#define I2S_MIC_SERIAL_CLOCK GPIO_NUM_22
#define I2S_MIC_LEFT_RIGHT_CLOCK GPIO_NUM_19
#define I2S_MIC_SERIAL_DATA GPIO_NUM_21

// Analog Microphone Settings - ADC1_CHANNEL_7 is GPIO35
// #define ADC_MIC_CHANNEL ADC1_CHANNEL_7

// speaker settings
#define I2S_SPEAKER_SERIAL_CLOCK GPIO_NUM_14
#define I2S_SPEAKER_LEFT_RIGHT_CLOCK GPIO_NUM_12
#define I2S_SPEAKER_SERIAL_DATA GPIO_NUM_27

// command recognition settings
#define COMMAND_RECOGNITION_ACCESS_KEY "PXXXXXIRRSTABXFQ7UIXPFRMC4L5"

cgreening, 2021-03-11:
I've wired it up locally as you have described:

https://imgur.com/a/xViIYDX

I think the next step would be to test if the microphone is definitely working. You can try the code from the esp_audio repo https://github.com/atomic14/esp32_audio

```
Average detection time 104ms
P(0.99): Here I am, brain the size of a planet...
Free ram after DetectWakeWord cleanup 96400
Free ram before connection 96400
Ready for action
Free ram after connection 46568
3 seconds has elapsed - finishing recognition request
Http status is 200 with content length of 660
I heard "tell me a joke"
Intent is Tell_joke
Free ram after request 95900
```

arijitaich, 2021-03-11:
OK. I tried testing with the esp32_audio package.  Here is the output:

[E][WiFiGeneric.cpp:739] hostByName(): DNS Failed for 0.0.0.0 **REPEATS**

cgreening, 2021-03-12:
You need to use the IP address of your computer, not 0.0.0.0.

The server listens on 0.0.0.0 - that just means that the server is listening on all the network interfaces of your computer. But the EPS32 needs to know the actual IP address of your computer.

arijitaich, 2021-03-12:
Ok. I am a little bit confused here. And thanks a ton for helping me out here.

I have created the node server in my PC and flashing the ESP from my laptop. 
Just confirming, you want me to add the IP address which would be something like 192.168.0.X to the ESP.
Am i right?

cgreening, 2021-03-12:
Put the IP address of the machine that's running the node server in the ESP32 code - so in your case that will be your PC not the laptop.

e.g:

#define I2S_SERVER_URL "http://192.168.0.5:5003/i2s_samples"

arijitaich, 2021-03-12:
I think my microphone is gone. I need to re-purchase it. The I2S.raw file was totally blank. Entirely. Only some beeps and distortion during unplugging the board from the USB. Also something i noticed. When i am placing the board to Laptop USB power, it is writing files to the server. The moment i plug it into a PSU module for 5v, it is not running.

arijitaich, 2021-03-15:
Hi, Got a new INMP441. Tested out with the wiring:

INMP441 --------------------------- ESPWROOM32
SD ---------------------------------- D21
VCC --------------------------------- 3V3
GND -------------------------------- GND
SCK --------------------------------- D22
WS ---------------------------------- D19
L/R ---------------------------------- GND

Uploading the output I2S.raw file. I think its a power issue i guess.
http://osmo.tech/core/workspace/MachDelivery/i2s.raw

cgreening, 2021-03-15:
Did you change the code to match those pin numbers?

In the the file `i2s_sampling/src/main.cpp` it's set up with these pins:

```
// i2s pins
i2s_pin_config_t i2sPins = {
    .bck_io_num = GPIO_NUM_32,
    .ws_io_num = GPIO_NUM_25,
    .data_out_num = I2S_PIN_NO_CHANGE,
    .data_in_num = GPIO_NUM_33};
```

For the wiring you've posted you will need to change it to:

```
// i2s pins
i2s_pin_config_t i2sPins = {
    .bck_io_num = GPIO_NUM_22,
    .ws_io_num = GPIO_NUM_19,
    .data_out_num = I2S_PIN_NO_CHANGE,
    .data_in_num = GPIO_NUM_21};
```

arijitaich, 2021-03-15:
Yes I Did. Here is the code i used:
i2s_pin_config_t i2sPins = {
    .bck_io_num = GPIO_NUM_22,
    .ws_io_num = GPIO_NUM_19,
    .data_out_num = I2S_PIN_NO_CHANGE,
    .data_in_num = GPIO_NUM_21};

arijitaich, 2021-03-15:
https://osmo.tech/core/workspace/MachDelivery/i2s_without_GND_to_LR.raw
This is the version without the GND connected to LR.

cgreening, 2021-03-15:
Let's double-check the wiring you have. I've got an INMP441 wired up in the same way and it's working.

I've put an image of the microphone and an image of the ESP32 here:

https://imgur.com/a/UrYkqat

|mic pin|wire|esp32|
|---|---|---|
|WS | yellow wire | D19|
|SCK | blue wire | D22|
|SD | orange wire | D21|
|L/R | dark green wire | GND|
|GND | dark green wire | GND
|VCC | red wire | 3V3|

arijitaich, 2021-03-15:
I tried implementing. It's not working. The output is still the same. Distorted.
When i am unplugging the Mic from the Esp, the distortions don't get recorded. But when i connect them and restart,
It is transmitting to the server but same distorted packets like the example link i have attached in my previous response.

Does this mean I won't be able to do it?

cgreening, 2021-03-16:
It's very strange - normally even if you are seeing noise and distortion there would still be some underlying audio signal. 

With the files you've sent there is no audio. This makes me think that either the microphone is faulty or the wiring it incorrect.

You can try the MAX4466 - you'll need to use the ADC to read data from it. You can try using the same sample code - it has support for ADC instead of I2S.

arijitaich, 2021-03-16:
**Here is my config.h:**
// WiFi credentials
#define WIFI_SSID "OTTO_2.4G"
#define WIFI_PSWD "Jhffcv1233"

// are you using an I2S microphone - comment this out if you want to use an analog mic and ADC input
// #define USE_I2S_MIC_INPUT

// I2S Microphone Settings

// Which channel is the I2S microphone on? I2S_CHANNEL_FMT_ONLY_LEFT or I2S_CHANNEL_FMT_ONLY_RIGHT
#define I2S_MIC_CHANNEL I2S_CHANNEL_FMT_ONLY_LEFT
// #define I2S_MIC_CHANNEL I2S_CHANNEL_FMT_ONLY_RIGHT
#define I2S_MIC_SERIAL_CLOCK GPIO_NUM_22
#define I2S_MIC_LEFT_RIGHT_CLOCK GPIO_NUM_19
#define I2S_MIC_SERIAL_DATA GPIO_NUM_21

// Analog Microphone Settings - ADC1_CHANNEL_7 is GPIO35
// #define ADC_MIC_CHANNEL ADC1_CHANNEL_7

// speaker settings
#define I2S_SPEAKER_SERIAL_CLOCK GPIO_NUM_14
#define I2S_SPEAKER_LEFT_RIGHT_CLOCK GPIO_NUM_12
#define I2S_SPEAKER_SERIAL_DATA GPIO_NUM_27

// command recognition settings
#define COMMAND_RECOGNITION_ACCESS_KEY "P5QMUSXFQ7UIXPFRMC4LMFV6IRRSTAB5"

**Here is the output after i commented out the lines.**
```
Compiling .pio\build\esp32doit-devkit-v1\src\state_machine\RecogniseCommandState.cpp.o
src/main.cpp: In function 'void setup()':
src/main.cpp:103:55: error: 'ADC_MIC_CHANNEL' was not declared in this scope
   I2SSampler *i2sSampler = new ADCSampler(ADC_UNIT_1, ADC_MIC_CHANNEL);
                                                       ^
src/main.cpp:121:46: error: 'i2s_sampler' was not declared in this scope
   Application *application = new Application(i2s_sampler, intent_processor, speaker, indicator_light);
                                              ^
*** [.pio\build\esp32doit-devkit-v1\src\main.cpp.o] Error 1
```

cgreening, 2021-03-16:
I've pushed up a fix for that. But I've honestly not been able to get it to work with an analogue microphone - in my setup there's simply too much noise.

I know people have got it working so it's worth trying.

cgreening, 2021-03-16:
What you can do is lower the detection threshold - line 59 of `DetectWakeWordState.cpp`

```
    if (output > 0.95)
```

You can make this much lower which will increase the sensitivity of the detection - but that comes at the cost of a lot more false positives.

e.g.

```
    if (output > 0.05)
```

You can also change line 62

```
        if (m_number_of_detections > 1)
```

To 

```
        if (m_number_of_detections > 0)
```

For even more sensitivity.

With these changes I'm able to get it work reasonably consistently - but it's very sensitive to random noises.

arijitaich, 2021-03-16:
I tried with 2 ESP32 boards and 2 microphones. (MAX4466 & INMP441). Only once the output seemed to read my voice. Attaching the file here: https://osmo.tech/core/workspace/MachDelivery/i2s_vocal.raw

I think the issue is the wiring.
Just a thought. It's cool if it's not ok with you.
Is there a possibility of a circuit board design where I can just send it to PCB way and they will send me the board where I can just embed the ESP32 board and start running?

arijitaich, 2021-03-17:
Also, Do i need to provide the resistors in the schema like you have? Currently i am not using any.

cgreening, 2021-03-17:
I don't have any resistors. Hopefully, this will help explain the wiring up.

https://oshwlab.com/chris_9044/alexa-wiring-example

arijitaich, 2021-03-18:
Is that a 20K Resistor between ESP32 **VIN** & MAX98357A I2S Amp **GAIN**
Also where is the connection for VIN & GND in MAX98357A I2S Amp?

cgreening, 2021-03-18:
I've updated the example wiring diagram to include the amplifier:

https://oshwlab.com/chris_9044/alexa-wiring-example

The VIN and GND are hidden under the board in the picture - the wiring example shows the details.

The resistor is optional and controls the gain of the amplifier - see here: https://learn.adafruit.com/adafruit-max98357-i2s-class-d-mono-amp

arijitaich, 2021-03-20:
When i am trying to convert it to Gerber Files to place an order at JLC, it says:
**This PCB has incompleted connections, do you want to check the connections?**

I press NO here to proceed but then the circuit that is showing at the end shows like the screenshot attached.
Since this is my first time with PCB printing over online, i think the circuit doesnt have connections on it, 
where as you have provided the connections.

What i want to know is will the connections be printed on the PCB even if the connections are not showing in the end preview?
Please let me know. I will initiate the order on the basis of the information provided.

Thanks.

Link: http://osmo.tech/core/workspace/MachDelivery/gerber.png
[image]

arijitaich, 2021-03-21:
The issue mentioned in the previous comment has been sorted. Now i know how to design this. 
I would like to wait until the PCB comes up and i perform a test successfully.
Once done, my issue would be resolved. I just hope this works,.
[image]

cgreening, 2021-03-21:
Don't order it yet! The schematic was only meant as a wiring guide, not to build a PCB from. I can see several errors in what you have done.

You have wired up the amplifier output which should be going to the speaker - I'm not sure why you've done that? Remove all the tracks going from the output of the amplifier.

arijitaich, 2021-03-21:
Thanks A Ton. You Are Awesome. Thank God you checked it out. I thought that too but then again thought, since EasyEDA knows more about these than me, hemce it must be something which is a mandate. These connections were pre provided when i converted the diagram into a PCB. I am sharing the link of my design:
https://oshwlab.com/a2.arijitaich/alexa-wiring-example

cgreening, 2021-03-21:
The footprint was supplied by someone was not correct - I've replaced it with a simple header.

Please have a look at the new schematic and the PCB layout here - https://oshwlab.com/chris_9044/alexa-wiring-example

Double-check the wiring on both the schematic and the PCB layout to the microphone and amplifier board. Make sure that it corresponds to `config.h`

Also, make sure that the pin layout for ESP32 dev board matches your board. These are all contributed by other people so there is guarantee that they are accurate.

You may want to modify the PCB to break out some of the ESP32 pins.

I can't guarantee that having a PCB will fix the issues you are seeing - I would prefer it if you could work out why it's not working on a breadboard.

cgreening, 2021-03-21:
I had a brief look at your schematic - I would swap the amplifier with a simple header as I have done - the user-contributed footprint is wrong and just confuses things.

Also, check that the ESP32 footprint matches your dev board. I've changed mine as the PINs in the PCB didn't match the board I have which is a generic dev board.

arijitaich, 2021-03-21:
I also want this to work in this stage but unfortunately its not working. I tried doing it on a bread board to eliminate chances of connection error but still the out is the same. Distorted.

Also, when i am testing out the microphone, it is throwing similar results. When i unplug the microphone, the entire track is empty. No sound gets recorded which means the INMP441 module is working but unfortunately it is not generating the desired output. 

I ll order one more microphone and test out the MAX4455 solution tomorrow.

arijitaich, 2021-03-22:
Thats the output when i commented out the I2S line as mentioned.
```
Building in release mode
Compiling .pio\build\esp32doit-devkit-v1\src\main.cpp.o
Compiling .pio\build\esp32doit-devkit-v1\src\state_machine\RecogniseCommandState.cpp.o
src/main.cpp: In function 'void setup()':
src/main.cpp:121:46: error: 'i2s_sampler' was not declared in this scope
   Application *application = new Application(i2s_sampler, intent_processor, speaker, indicator_light);
                                              ^
*** [.pio\build\esp32doit-devkit-v1\src\main.cpp.o] Error 1
```

arijitaich, 2021-03-25:
Hi, I ordered 3 brand new INMP441 units & 1 ESP32 Dev Board.
Connected them on a bread board and tested all connections using a multimeter.
Then tried running the ESP32_audio project for I2S testing. The recording which is being sent to the server does not have any difference. Its all distorted. Sounds like someone trying to send SOS messages.

So its not a problem of connectors. Neither it is for the board. Nor is it your coding since you have already shown them on your end that they are working.

If all of the above options are obsolete, then what is the issue? 
I think it will only work with the mic breakout board you are using, and frankly that is not available here in India.

Hence, i have purchased 2 breakout board which you have created from tindie. I am adamant to make this work.
Either it will be proved that this thing works and can be replicated elsewhere as a project or not.
But i think this will act as a closure for me.

cgreening, 2021-03-25:
Can you put the code you are using and the wiring diagram on GitHub so I can double-check them?

Assuming the wiring is correct then the only other thing I can think of is the power coming to the microphone - but you should see some audio coming through.

But let's double check the i2s sampling code you are using and the wiring you have.

arijitaich, 2021-03-25:
I2S Sampling Git:
https://github.com/arijitaich/i2s_sampling

Wiring Image:
http://osmo.tech/core/workspace/MachDelivery/20210325_173745.jpg
http://osmo.tech/core/workspace/MachDelivery/20210325_173734.jpg
http://osmo.tech/core/workspace/MachDelivery/20210325_173754.jpg

Wiring Diagram:
3v3 **ESP32** --> VDD **INMP441**
GND **ESP32** --> GND & L/R **INMP441**
D5 **ESP32** --> SCK **INMP441**
D18 **ESP32** --> SD **INMP441**
D19 **ESP32** --> WS **INMP441**

Please let me know if you need anymore information.

cgreening, 2021-03-25:
Can you make https://github.com/arijitaich/i2s_sampling public? I think it must be set to private at the mment.

arijitaich, 2021-03-25:
Done. Sorry. That was an error on m part. Now it's corrected to public.

cgreening, 2021-03-25:
Well, everything looks correct - can you upload some more sample audio? I can't see any reason for it not to work.

The only thing I can see that is a bit strange is the header pins - normally you'd have the yellow part on the bottom so the pins might be a bit short - are they definitely making contact with the breadboard? Does the multimeter show continuity from the header pin to the ESP32 pin?

https://imgur.com/a/7RxOBMj

arijitaich, 2021-03-25:
The pins are not soldered. I have used the pins to embed into the breadboard.
Yes, I have checked with the continuity option using the Multimeter.

**Here is the audio output sample you asked for:** 
Sample 1:  http://osmo.tech/core/workspace/MachDelivery/i2s_25_3_20.raw
Sample 2:  http://osmo.tech/core/workspace/MachDelivery/i2s_25_3_20_2.raw
Sample 3:  http://osmo.tech/core/workspace/MachDelivery/i2s_25_3_20_3.raw

**Here is the video showing Sample 1 collection demonstration:**
https://www.youtube.com/watch?v=IJ-KTmYksMM

cgreening, 2021-03-25:
Ah, unsoldered pins will cause a lot of issues - it's likely that at least one of them won't be making a good connection to the microphone board.

arijitaich, 2021-03-25:
Alright. I do not have a pencil point soldering rod. I have a thick one which wont make good connections here. 
I will purchase it tomorrow and try it out. 
In the meantime, have you received the orders for those microphones? If you have let me know, when you have dispatched them since i do not have any option to track it.
The best way is to know when it left its starting point. Sorry to bother and thanks for staying with me.

cgreening, 2021-03-25:
They should be sent tomorrow morning - it will take 6-7 days to arrive. Do you want me to solder the header pins on?

arijitaich, 2021-03-25:
Can you? That would be of immense humungous help.

arijitaich, 2021-03-26:
Tested out the INMP41 after soldering the heads of the pins. Output still the same. Distorted. NO vocals whatsoever.

**Here is the sample audio:**
http://osmo.tech/core/workspace/MachDelivery/i2s_26_3_20_1.raw
http://osmo.tech/core/workspace/MachDelivery/i2s_26_3_20_2.raw

**Here are the pictures of the soldered INMP441:**
http://osmo.tech/core/workspace/MachDelivery/20210326_134624.jpg
http://osmo.tech/core/workspace/MachDelivery/20210326_134706.jpg

Also, before initiating the test, i double-checked the connections using the continuity option of the multimeter.

cgreening, 2021-03-26:
I'm afraid I'm completely stumped.

The microphone boards will be sent this morning - I've assembled and tested them so they should definitely work.

arijitaich, 2021-03-26:
Great! I hope they work. If they don't I'll try out printing the PCB you had sent me.
If both the options don't work, then I won't give it any more tries.

[3 more replies]

closed 2021-11-16 by cgreening