atomic14/diy-alexa #13: InvalidArgumentError: unknown file type: speech_data\backward\0165e0e8_nohash_0.wav [Op:IO>AudioReadableInit]

edwardcorke, 2021-03-14:
Hi there,

Firstly, what a great project and thank you for all the information you have provided! 

I've been implementing my own version for the firmware, but am trying to use your jupyter notebook for preprocessing the dataset. I downloaded the same dataset, extracted the files using the same command and ran your notebook 'Generate Training Data.ipynb' but get the error:

InvalidArgumentError: unknown file type: speech_data\backward\0165e0e8_nohash_0.wav [Op:IO>AudioReadableInit]

Is there anything you could recommend to solve this or provide you with more information needed?

Thanks a lot!
Edward

cgreening, 2021-03-14:
Hi Edward,

Thanks for your kind words - I really appreciate it.

Are you running on a mac, windows or Linux machine? It sounds like something might be missing that is required to read in the WAV files.

Somewhat annoyingly it's working on my machine, but I can spin up a virtual machine and match what you have.

Cheers
Chris

edwardcorke, 2021-03-14:
Hi Chris,

I am using Windows 10 and Python 3.7.9. Interestingly, I can't pip install pyaudio. I didn't think this would be a problem as it's not an import for 'Generate Training Data.ipynb', but maybe I am wrong.

Thank you for your quick response,
Edward

cgreening, 2021-03-14:
Hi Edward,

I think that might be the problem - I think pyaudio is required by TensorFlow to read the wav files.

This might help - though I'm not sure as I don't use windows day to day.

https://stackoverflow.com/questions/52283840/i-cant-install-pyaudio-on-windows-how-to-solve-error-microsoft-visual-c-14

arijitaich, 2021-03-15:
It works on Windows 10 with Python 3.8. Do not go ahead with Python 3.9. You'll face multiple other problems.

edwardcorke, 2021-03-30:
Hi guys,

Thank you both for providing useful information. You were correct and I managed to get it working running a Python 3.6 container.

Thanks again,
Edward

closed 2021-03-31 by edwardcorke