atomic14/diy-alexa #16: Custom Wake Words Recognition

mdsuffian, 2021-04-26:
Hello, I try to made my custom waking words using my own trained voice but it was not working.
the end result of generating the testing data shows empty results of words "testing"
[image]

I did try Records 10 samples of words "testing" in audacity and export the results to 256 kb/s. 16.0kHz , 16 bits , 1 channel , PCM ( Little / Signed)  .  

The "Testing" folder was created inside the model\speech_data. and all of the wave files exported placed under "diy-alexa-master\model\speech_data\testing" folder.

In Generate Training Data.ipynb , I change the source as following :

**_1. Adding "Testing" to the words array._**

# list of folders we want to process in the speech_data folder
from tensorflow.python.ops import gen_audio_ops as audio_ops
words = [
    'backward',
    'bed',
    'bird',
    'cat',
    'dog',
    'down',
    'eight',
    'five',
    'follow',
    'forward',
    'four',
    'go',
    'happy',
    'house',
    'learn',
    'left',
    'marvin',
    'nine',
    'no',
    'off',
    'on',
    'one',
    'right',
    'seven',
    'sheila',
    'six',
    'stop',
    'testing',
    'three',
    'tree',
    'two',
    'up',
    'visual',
    'wow',
    'yes',
    'zero',
    '_background',
]

**_2. Replace the words "marvin" to word "testing" in the following code_**
# process all the words and all the files
for word in tqdm(words, desc="Processing words"):
    if '_' not in word:
        # add more examples of marvin to balance our training set
        # repeat = 70 if word == 'testing' else 1
        repeat = 70 if word == 'testing' else 1
        process_word(word, repeat=repeat)
    
print(len(train), len(test), len(validate))

**_3. Last , I added the following code to the end of the code_** 

word_index = words.index("testing")

X_testing = np.array(X_train)[np.array(Y_train) == word_index]
Y_testing = np.array(Y_train)[np.array(Y_train) == word_index]
plot_images2(X_testing[:20], IMG_WIDTH, IMG_HEIGHT)
print(Y_testing[:20])

cgreening, 2021-04-26:
Apologies, I can see what the issue is.

All the samples need to be exactly 1 second long.

The function `def is_valid_file(file_name):` checks that there are 16000 samples and rejects files that are not the correct length.

This length limit is due to the way the model works - it treats the spectrogram like a 2D image, so can't deal with longer or shorter samples.

There's a couple of options:

Make sure all your files are exactly 1 second in length
Or:
Modify the code so that it truncates the audio file to 1 second and only rejects files that are too short. You would need to make sure you have at least 1 second of data in the file.

You can modify the `def is_correct_length(audio, expected_length):` to accept files that are `>=16000` samples.

And modify the `def process_file(file_path):` function to something like:

```
def process_file(file_path):
    # load the audio file
    audio_tensor = tfio.audio.AudioIOTensor(file_path)[0:16000]
```

I think that will work - I've not tested it though.

mdsuffian, 2021-04-26:
Brilliant ! it's now capturing perfectly !   : ) 
[image]

I trimmed all sound to exact 1 seconds , no more or less.

closed 2021-04-26 by mdsuffian