atomic14/diy-alexa #20: IDF port advice?

WifWaf, 2021-08-07:
Hi atomic, many thanks for the firmware.

Apologies for opening an issue, as it's not an issue with your library.

A few months back I ported your firmware to IDF. While it works, I found when calling [invoke()](https://github.com/WifWaf/diy-alexa/blob/9f43089bc7b21d8f2ebc7fbaeb4752c434aba4b9/firmware/components/neural_network/neural_network.cc#L46) the response time was very slow (I think it was roughly 2.5 times the original). 

I measured the time taken for code execution and the slow response came down to calling the TensorFlow function.

I was wondering if you had any insight into why this might be? Originally I thought it might be an issue with the C-linkage, however, I ran my code through Arduino too as .cpp.

If not, no worries; I was planning to wait for the C implementation for TensorFlow Micro 👍 . Cheers,

cgreening, 2021-08-07:
Could be one of two things.

The most likely will be the build flags that are being used. In my code I used the `-Ofast` flag which compiles with all the optimisations turned on and also has slightly less accurate maths - see here for the different options - https://gcc.gnu.org/onlinedocs/gcc-4.1.0/gcc/Optimize-Options.html

You may be building in debug mode or with less aggressive optimisations - it's most likely that you are building in debug mode as I think that might be the default (though I'm not sure).

The other possibility is the core you are running on - I think I pinned the recognition task to a one of the CPU cores so that it didn't run on the same core as the WiFi and the audio sampling - but I don't think it made a dramatic difference.

WifWaf, 2021-08-07:
Unfortunately, so far, the build flags have made little difference (nor pining the task/raising priority) - though I had actually set it compile for size optimisation and forgot...

No doubt I'm missing something somewhere else too; I'll have a better read over the build flags tomorrow.

Much appreciate the help (I'll close the issue :)).

closed 2021-08-07 by WifWaf