atomic14/diy-alexa #23: i have a question regarding this alexa project

Anugaradhawa, 2021-11-17:
I have a question regarding this project  (https://github.com/atomic14/diy-alexa ), how do I import background audio files to the jupyter notebook? plz reply for this questions

cgreening, 2021-12-03:
Can you explain your question in a bit more detail? I'm not sure what it means.

Anugaradhawa, 2021-12-03:
> Can you explain your question in a bit more detail? I'm not sure what it means.

ok first of all thanks for your reply I waiting for your reply for a long period, my first question is am I need to import background noises to Jupyter notebook friend?? I waiting for your quick reply

cgreening, 2021-12-03:
I'm afraid I still don't really understand your question. There is background noise included in the command dataset that you can download as in the instructions in the notebook.

If you just want to run the project then the model is already trained and included in the firmware. You only need to train the model if you want to change the wake word.

```
Download data set
Download from: https://storage.cloud.google.com/download.tensorflow.org/data/speech_commands_v0.02.tar.gz - approx 2.3 GB

And then run

tar -xzf data_speech_commands_v0.02.tar.gz -C speech_data
```

Anugaradhawa, 2021-12-03:
> I'm afraid I still don't really understand your question. There is background noise included in the command dataset that you can download as in the instructions in the notebook.
>
> If you just want to [...]

can you send your phone number, then I can send screenshots of errors in jupyter notebook friend . it's help for understand my question

cgreening, 2021-12-03:
You can attach screenshots here. This is the best place to manage issues as it means everyone else can get value from the solutions. You can also copy and paste the errors from the jupyter notebook.

Are you trying to train the model on new data?

Anugaradhawa, 2021-12-03:
ok friend

Anugaradhawa, 2021-12-03:
now in my country10.37 pm, therefore I am going to sleep thanks for your replies I will be send my issues on tomorrow.

Anugaradhawa, 2021-12-04:
> You can attach screenshots here. This is the best place to manage issues as it means everyone else can get value from the solutions. You can also copy and paste the errors from the jupyter notebook.
> 
> Are you trying to train the model on new data?

I try to train  model but it some errors

Anugaradhawa, 2021-12-04:
[image]
i try to generate training data but it says "FileNotFound" please tell me what I do for it

cgreening, 2021-12-04:
If you want to train the model from scratch then you need to generate the training data by following the steps in the "Generate Training Data" notebook.

But as I said before, there is a fully trained model already in the firmware. You only need to train a new model if you want to have a new wake work.

Anugaradhawa, 2021-12-04:
yes I want to make a new wake word

Anugaradhawa, 2021-12-04:
> If you want to train the model from scratch then you need to generate the training data by following the steps in the "Generate Training Data" notebook.
>
> But as I said before, there is a fully [...]

what notebook?? jupyter notebook ??

cgreening, 2021-12-04:
Using a new wake word is a very advanced thing to try and do.

You will need to collect a lot of 1 second long samples of people saying your wake word and then regenerate the training data using the generate training data notebook.

Anugaradhawa, 2021-12-04:
OK then I'll use the wake word you created

Anugaradhawa, 2021-12-04:
am I need to use jupyter notebook for this project??

cgreening, 2021-12-04:
If you just want to use the pre-trained model then you don't need to use the notebooks. You just need to build the firmware and run it on your ESP32.

cgreening, 2021-12-04:
You will need to have some experience of using the esp32 with PlatformIO. If you have not done this before then try a few simple projects first.

Anugaradhawa, 2021-12-04:
ok 

> You will need to have some experience of using the esp32 with PlatformIO. If you have not done this before then try a few simple projects first.

 I didn't use esp 32 before with platform.io, ok I will practice some projects using ESP 32 . where I found some tutorials??

Anugaradhawa, 2021-12-04:
> If you just want to use the pre-trained model then you don't need to use the notebooks. You just need to build the firmware and run it on your ESP32.

how I build firmware for esp 32, Are you did some tutorials to make a firmware

Anugaradhawa, 2021-12-04:
> If you just want to use the pre-trained model then you don't need to use the notebooks. You just need to build the firmware and run it on your ESP32.

can I use firmware that you create??

cgreening, 2021-12-04:
This might be a good starting point: https://randomnerdtutorials.com/vs-code-platformio-ide-esp32-esp8266-arduino/

Anugaradhawa, 2021-12-04:
> This might be a good starting point: https://randomnerdtutorials.com/vs-code-platformio-ide-esp32-esp8266-arduino/
[image]

ok thanks I will try this project on esp 32, but can you give a reason for this error??

cgreening, 2021-12-04:
You haven't generated any training data so the files do not exist. If you want to train the model then you need to first run the notebook that generates the training data.

Anugaradhawa, 2021-12-04:
> You haven't generated any training data so the files do not exist. If you want to train the model then you need to first run the notebook that generates the training data.

How i generate training data?? , Am i need background audio for making training data??

cgreening, 2021-12-04:
Have you tried running the generate training data notebook and have followed the instructions in that notebook?

Anugaradhawa, 2021-12-05:
> Have you tried running the generate training data notebook and have followed the instructions in that notebook?

what notebook?? I have tried to generate training data on jupyter notebook but it's come some errors

Anugaradhawa, 2021-12-05:
[image]

can you tell me what software you use in this picture ??? apple terminal or other software??

cgreening, 2021-12-05:
This is just a command to play an audio file - you can use any command you want. The one I'm using is called `play` and comes as part of the operating system on the mac (and on Linux) as well.

You can use Audactity - https://www.audacityteam.org/ or whatever you have on windows. You can also install WSL (https://docs.microsoft.com/en-us/windows/wsl/install) on windows which will give you a Linux shell.

You are still trying to generate training data even though you said you didn't need to. I can't help you if you don't tell me what the errors are when you try and do it.

Anugaradhawa, 2021-12-05:
[image]

error is "FileNotFound"  what i do for this

cgreening, 2021-12-05:
I've already explained this. You have not generated the training data. You need to run the notebook called: "Generate Training Data.ipynb"

Anugaradhawa, 2021-12-05:
> I've already explained this. You have not generated the training data. You need to run the notebook called: "Generate Training Data.ipynb"

sorry sorry I forget it

Anugaradhawa, 2021-12-05:
[image]

in this code part there have function names in "get_file" , That mean am I need to add some recorded background audio ??

Anugaradhawa, 2021-12-05:
[image]

when I paste your curl command on command prompt in windows it's come "code": "no-auth", "error": "Bad auth, check token/params" what I do for this??

cgreening, 2021-12-06:
On windows, you will need to change the single quotes to double quotes. But you also need to sign up with wit.ai and get your access token to replace the "XXX". 

In the instructions:

```
You can get the exact values for the Authorization header and the URL from the settings page of your Wit.ai application.
```

cgreening, 2021-12-06:
If you can open new issues for problems you have that are specific to the issue you are facing it will be easier for people to help you. Give the issue a meaningful title and make sure you have followed the instructions.

closed 2021-12-06 by cgreening

Anugaradhawa, 2021-12-06:
anyone not replies to my issue