atomic14/diy-alexa #28: problem releated to the code

Anugaradhawa, 2021-12-07:
[image]

In the middle of this picture, the code has a function named "get_files" highlighted in purple clour, Are you use this function to import the background audio files to the jupyter notebook ??

Anugaradhawa, 2021-12-07:
[image]

but in jupyter notebook does not highlight the get_files function

cgreening, 2021-12-07:
You need to run all the cells in the notebook. You can't simply copy a section of code out of the notebook and expect it to work.

What are you trying to do?

Anugaradhawa, 2021-12-07:
> What are you trying to do?

Why, Don't want to do that??

cgreening, 2021-12-07:
You should clone the repository:

`git clone git@github.com:atomic14/diy-alexa.git`

And you'll get the complete notebooks. There is a readme file in the `model` folder where the notebooks are.

You can't just copy bits out of the notebook.

cgreening, 2021-12-07:
But, as I've said in your other issues, you don't need to run the notebooks if you just want to use the firmware with the current wake word. So I'm not sure what you are trying to do.

Anugaradhawa, 2021-12-07:
> But, as I've said in your other issues, you don't need to run the notebooks if you just want to use the firmware with the current wake word. So I'm not sure what you are trying to do.

Ahh ok ok i use the current Wake word for my project, I asked to know that

Anugaradhawa, 2021-12-08:
When I want to make a new wake word am I need to import the background audio files to the jupyter notebook??

cgreening, 2021-12-08:
You need to look at the notebooks in the GitHub repository and read the code that generates the training dataset. I'm afraid your questions aren't really making any sense to me.

Anugaradhawa, 2021-12-08:
> You need to look at the notebooks in the GitHub repository and read the code that generates the training dataset. 

ok ok I already looked the code in GitHub but my issue is are you import background audio files to the jupyter notebook, I want a "yes" or "no" answer

Anugaradhawa, 2021-12-08:
Are we need some background audio files for made training data??

cgreening, 2021-12-08:
Ok, I think I understand what you are asking. There are samples of background noise included in the dataset that you download from here: https://storage.cloud.google.com/download.tensorflow.org/data/speech_commands_v0.02.tar.gz

Follow the instructions in the notebook and extract the file to the speech_data folder.

cgreening, 2021-12-08:
If you don't want to use background noise then you can comment out that part of the code.

closed 2021-12-08 by cgreening

Anugaradhawa, 2021-12-08:
> Ok, I think I understand what you are asking. There are samples of background noise included in the dataset that you download from here: [...]

Ok tell me why are you use this background audio files??  For code or for training data ??