atomic14/diy-alexa #30: Background audio files

Anugaradhawa, 2021-12-08:
> Ok, I think I understand what you are asking. There are samples of background noise included in the dataset that you download from here: https://storage.cloud.google.com/download.tensorflow.org/data/speech_commands_v0.02.tar.gz
> 
> Follow the instructions in the notebook and extract the file to the speech_data folder.

Ok tell me why are you use this background audio files??  For code or for training data ??

Anugaradhawa, 2021-12-09:
ok i want to know are you use background audio files for training data??

cgreening, 2021-12-09:
You want to train the neural network to detect your wake word. To do this it needs example audio files containing your wake word and example audio files that don't contain your wake word.

You also want it to be able to detect your wake word even when there is background noise. Or at least to be reasonably robust when there is background noise.

The background noise is used in the training of the neural network.

Anugaradhawa, 2021-12-09:
> You want to train the neural network to detect your wake word. To do this it needs example audio files containing your wake word and example audio files that don't contain your wake word. These counter examples can be other words and background noise.
> 
> You also want it to be able to detect your wake word even when there is background noise. Or at least to be reasonably robust when there is background noise.
> 
> The background noise is used in the training of the neural network.

ok how i training the neural network ?? using a jupyter notebook??

cgreening, 2021-12-09:
You can use the jupyter notebooks in the repository, but if you want to use different training data then you will need to modify the code.

There are some good tutorials on the TensorFlow website that you can use to learn how it all works, but you are in for a lot work!

https://www.tensorflow.org/overview

Anugaradhawa, 2021-12-09:
[image]

now I run the training data notebook but it'll occur an error "Processing problem noise" what is the reason??

Anugaradhawa, 2021-12-09:
[image]

how do you add the audio sample to the training data??

cgreening, 2021-12-09:
I recorded audio using my computer and built-in microphone to record additional background sounds.

You can comment out the lines that deal with the "mar" sounds if you don't need it.

Anugaradhawa, 2021-12-10:
> I recorded audio using my computer and built-in microphone to record additional background sounds.
> 
> You can comment out the lines that deal with the "mar" sounds if you don't need it.

I ask how you add audio sample to the Jupyter noteboo

cgreening, 2021-12-10:
I'm afraid the question doesn't make sense to me.

Are you asking how to read audio data into Python?

I think that you should follow some tutorials using jupyter notebook and tensorflow to learn more. The link I sent previously: https://www.tensorflow.org/overview is a good place to start.

There is also a good set of videos here: that will be very helpful for you to watch: https://www.youtube.com/playlist?list=PL-wATfeyAMNrtbkCNsLcpoAyBBRJZVlnf

Anugaradhawa, 2021-12-10:
Ok i will watch the tutorial but i want to know how you add audio

Anugaradhawa, 2021-12-10:
If you can't understand my question, let me know and I will try to explain my question a bit detail

cgreening, 2021-12-10:
I don't understand your question.

Anugaradhawa, 2021-12-11:
ok first tell me are you add audio samples to the jupyter notebook, I want answer in "yes" or "no" only

Anugaradhawa, 2021-12-11:
> ok first tell me are you add audio samples to the jupyter notebook, I want answer in "yes" or "no" only

First answer this question , it's help you for understanding my idea

Anugaradhawa, 2021-12-12:
[image]
 
How you add random sample of background??

Anugaradhawa, 2021-12-12:
[image]
 
in your youtube vedio you tell " we also add in some random sample of background noise " I want to know only how you ass background samples ??

Anugaradhawa, 2021-12-12:
[image]

Are we need to add this data_speech_commands folder to the notebook then it will run through all these samples and output files?

Anugaradhawa, 2021-12-12:
Whya are you didn't response for my question ??

upendramadhava, 2021-12-12:
> [image]
> 
> Are we need to add this data_speech_commands folder to the notebook then it will run through all these samples and output files?

I think you need to add command data file to the jupyter notebook friend

upendramadhava, 2021-12-12:
> [image]
> 
> Are we need to add this data_speech_commands folder to the notebook then it will run through all these samples and output files?

I also have this question

cgreening, 2021-12-12:
> Whya are you didn't response for my question 😔??

I have other things to do as well as support GitHub projects!

Some of your questions can be answered by reading and understanding the code in the notebook.

You download the speech commands dataset and uncompress it to the `speech_data` folder. Follow the instructions at the top of the notebook:

```
Download data set
Download from: https://storage.cloud.google.com/download.tensorflow.org/data/speech_commands_v0.02.tar.gz - approx 2.3 GB

And then run

tar -xzf data_speech_commands_v0.02.tar.gz -C speech_data
```

This will give you a folder containing a list of words along with sample audio for each word.

The code runs over the list of words and extracts features for each word. If the word is the wake word - in our case "marvin" then we repeat it multiple times so that we have a balanced data set - approximately an equal number of wake words examples vs non-wakeword examples.

I've added an additional folder to the `speech_data` folder with some background noise audio. Just create a folder in the `speech_data` folder called `_background_noise_` and put some recordings there of background noise.

If you don't have any background noise then comment out these lines:

```
for file_name in tqdm(get_files('_background_noise_'), desc="Processing Background Noise"):
    process_background(file_name, words.index("_background"))
```

The same is true for `_problem_noise_`. Create a folder called `_problem_noise_` in the speech data folder and put audio that seems to confuse the neural network there.

Once again if you don't have any problem noise samples then comment out these lines:

```
for file_name in tqdm(get_files("_problem_noise_"), desc="Processing problem noise"):
    process_problem_noise(file_name, words.index("_background"))
```

The same is the case for the `_mar_sounds_`. I recorded a bunch of words that sound like "marvin" to help the neural network ignore similar words.

Comment out these lines if you don't have these samples:

```
for file_name in tqdm(get_files("_mar_sounds_"), desc="Processing problem noise"):
    process_mar_sounds(file_name, words.index("_background"))
```

Anugaradhawa, 2021-12-12:
> I have other things to do as well as support GitHub projects!

Ok i know 🙂 i was thinked you still can't understand my question

Anugaradhawa, 2021-12-12:
> process_background(file_name

That's mean when i haven't background audio files then can i use this function "for file_name in tqdm(get_files('_background_noise_'), desc="Processing Background Noise"):
    process_background(file_name, words.index("_background"))"  ??

Anugaradhawa, 2021-12-12:
> And then run
> 
> tar -xzf data_speech_commands_v0.02.tar.gz -C speech_data

Where i run this ?? on command prompt ??

Anugaradhawa, 2021-12-12:
> If you don't have any background noise then comment out these lines:
> 
> ```
> for file_name in tqdm(get_files('_background_noise_'), desc="Processing Background Noise"):
>     process_background(file_name, words.index("_background"))
> ```
> 

When I have background files then how i add that background files to the code ??

cgreening, 2021-12-13:
> > process_background(file_name
> 
> That's mean when i haven't background audio files then can i use this function " for file_name in tqdm(get_files('_background_noise_'), desc="Processing Background Noise"): process_background(file_name, words.index("_background")) " ??
> 
> > And then run
> > tar -xzf data_speech_commands_v0.02.tar.gz -C speech_data
> 
> Where i run this ?? on command prompt ??

You run this on the command prompt in the model folder. I don't think this will work in the normal windows command prompt. You would be better off installing WSL so that you have the full set of UNIX commands to work with.

cgreening, 2021-12-13:
> _background_noise_

Put the 

> When I have background files then how i add that background files to the code ??

Create a folder called `background_noise` in the `speech_data` folder and put the files there.

Anugaradhawa, 2021-12-13:
> Create a folder called `background_noise` in the `speech_data` folder and put the files there.

ok thanks

Anugaradhawa, 2021-12-13:
> You run this on the command prompt in the model folder. I don't think this will work in the normal windows command prompt. You would be better off installing WSL so that you have the full set of [...]

ok

Anugaradhawa, 2021-12-13:
> 

Are this step is essential for this project?? can I skip this step??

Anugaradhawa, 2021-12-13:
> Put the
>
>
> Create a folder called `background_noise` in the `speech_data` folder and put the files there.

Are this step is essential for this project?? can I skip this step??

Anugaradhawa, 2021-12-13:
[image]

 I run this "Generate Training Data
" notebook but it's occur a error "Processing Background Noise:
0/0 [00:00<?, ?it/s]
0 0 0" what is the reason??

Anugaradhawa, 2021-12-13:
> Create a folder called `background_noise` in the `speech_data` folder and put the files there.

Anugaradhawa, 2021-12-13:
Why use this code "> > > ```
> > > for file_name in tqdm(get_files('_background_noise_'), desc="Processing Background Noise"):
> > >     process_background(file_name, words.index("_background"))"

upendramadhava, 2021-12-13:
> You run this on the command prompt in the model folder. I don't think this will work in the normal windows command prompt. You would be better off installing WSL so that you have the full set of [...]

Yes it's not work correctly in Windows command prompt

Anugaradhawa, 2021-12-13:
> Yes it's not work correctly in Windows command prompt,we can also use Ubuntu software on windows for this work

ok thanks friend

Anugaradhawa, 2021-12-13:
[image]
your
[image]
my 
watch the differance of these screenshots

cgreening, 2021-12-13:
Why do you think it is happening? I don't know what is happening on your machine you need to debug the code and work out what the problem is.

It seems likely that it can't find the audio files. Have you downloaded the speech data and extracted it? Is it in the right place? Where is the code looking for the files?

Anugaradhawa, 2021-12-13:
> It seems likely that it can't find the audio files. Have you downloaded the speech data and extracted it? Is it in the right place? Where is the code looking for the files?

i also thinked that

Anugaradhawa, 2021-12-13:
> Why do you think it is happening? I don't know what is happening on your machine you need to debug the code and work out what the problem is.
> 
> It seems likely that it can't find the audio files. Have you downloaded the speech data and extracted it? Is it in the right place?

I also think  it's can't find the audio files.yes i downloaded the speech data and extract in right click on zip file and extract it.i not place anywhere that file. I  just only download and extract it

cgreening, 2021-12-13:
You need to create a folder called `speech_data` in the model folder (where you are running the notebooks) and copy the extracted files into that folder.

It expects to have all the words from the commands dataset under the `speech_data` folder.

Anugaradhawa, 2021-12-13:
> (where you are running the notebooks) 
I only open a notebook in anaconda and run it normally

Anugaradhawa, 2021-12-13:
> You need to create a folder called `speech_data` in the model folder (where you are running the notebooks) and copy the extracted files into that folder

How i copied speech data folder to model folder??

Anugaradhawa, 2021-12-13:
I put the speech_data file to the model folder now it's working pretty thanks you solve my issue thank you

closed 2021-12-13 by Anugaradhawa