atomic14/diy-alexa #31: General esp32 question (spiffs)

fluxxion82, 2021-12-12:
so I don't think this is actually an issue with your code. It's more of a general esp32 question that I'm hoping you can give some insight on.

I've got the project running and working except I'm having trouble playing the .wav files. In one of the projects I was running before I was getting a 'SPIFFS failed to mount' error, but when I run this project, I don't get that error. However, I get some errors when trying to load the wave files:

```
ERROR: bit depth 16379 is not supported please use 16 bit signed integer
ERROR: sample rate 200 is not supported please us 16KHz
fmt_chunk_size=0, audio_format=0, num_channels=0, sample_rate=200, sample_alignment=7984, bit_depth=16379, data_bytes=1073469932
```

I get the exact same messages for every wav file that the app attempts to open. I've called SPIFFS.format() but that doesn't do much to help. I am getting a value returned when I check for the total size (~1.3MB). I thought the esp32 has 4MB of SPI flash, and I don't see this spiff size in the spiffs config anywhere so I haven't verified if this is the correct value yet. I've tried running the project on two different dev boards so far but both behave the same way. 

I just tried out this esp32 data uploader for the arduino ide, was able to upload one of the joke files and the app successfully played the file. It seems like it's only able to play the file once though. Maybe it's a platform.io issue?

Anyways, if you have any insight, I would love to hear it. thanks.

cgreening, 2021-12-13:
Sounds like the SPIFFS filesystem has not been uploaded. This is done separately from uploading the firmware and it's slightly hidden away on PlatformIO. Pretty much impossible to explain in words so I made a video :)

https://youtu.be/Pxpg9eZLoBI

fluxxion82, 2021-12-14:
Awesome, that did the trick! Still working through your videos :)

closed 2021-12-14 by fluxxion82

cgreening, 2021-12-14:
Great! It's really annoying that they've hidden the upload file system as it's used a lot!

fluxxion82, 2021-12-18:
I have another random question...how would you add volume control to the project? I've attached a pot to one of the ADC pins and was trying to read the analog value in the WavReader getFrames function and then trying to apply the value to the audio data but it seems that having just the analog read call (without using the value) stops the audio from being played. I don't see any errors. Any thoughts? Thanks.