atomic14/diy-alexa #33: Generate Training Data

Anugaradhawa, 2021-12-13:
when I make a new wake word using "Generate Training Data" then How is there a connection between code and Generate Training Data in this project ??

cgreening, 2021-12-13:
If you look in both the `Train Model.ipynb` and `Generate Training Data.ipynb` you will see that there is a list of words.

`Generate Training Data` creates sets of features for all the audio samples for each word in the list and tags them with the index of the word in the list.

So for example `backward` = 0, `bed` = 1, `bird` = 2

The `Train Model` code marks any features that are not tagged with the same index as the word `marvin` as negative cases and all the features tagged with the same index as the word `marvin` as positive cases.

It then trains a binary classifier against this data.

In the firmware we feed the audio data to the trained model and it returns a probability that the word `marvin` appears in the audio data.

```
# List of the words in categorical order
words = [
    'backward',
    'bed',
    'bird',
    'cat',
    'dog',
    'down',
    'eight',
    'five',
    'follow',
    'forward',
    'four',
    'go',
    'happy',
    'house',
    'learn',
    'left',
    'marvin',
    'nine',
    'no',
    'off',
    'on',
    'one',
    'right',
    'seven',
    'sheila',
    'six',
    'stop',
    'three',
    'tree',
    'two',
    'up',
    'visual',
    'wow',
    'yes',
    'zero',
    '_background',
]
```

Anugaradhawa, 2021-12-13:
thanks i understand this

cgreening, 2021-12-13:
You need to collect a lot of samples of your wake word all of them 1 second long and you need to make sue you use 16KHz sampling rate.

Create a folder and in the `speech_data` directory and put your samples in there. Copy the name of the folder and put it in the list of words in both notebooks. When the notebook uses "marvin" change it to your wake words.

In the generate training data it repeats the samples for marvin many times to balance the dataset.

In the train model dataset it marks all the marvin words as the positive case.

You will need to learn Python to modify the notebooks. Please read and understand the code. I'm afraid I cannot do your project for you.

Anugaradhawa, 2021-12-14:
>  Please read and understand the code. I'm afraid I cannot do your project for you.
ok ok I try to make my own wake word thanks for helping for my project you

Anugaradhawa, 2021-12-14:
>  Please read and understand the code. I'm afraid I cannot do your project for you.

ok ok I try to make my own wake word thanks for helping for my project

closed 2021-12-14 by Anugaradhawa