atomic14/diy-alexa #34: platform io (error)

Anugaradhawa, 2021-12-14:
[image]

I install the visual studio code and download the platform io extension but its home page didn't open I wait a long time to open it. whats the reason you think ??

cgreening, 2021-12-14:
Do you see any details of you follow the instructions where it says it failed?

You will probably be able to get help here: https://community.platformio.org

Anugaradhawa, 2021-12-14:
> Do you see any details of you follow the instructions where it says it failed?
> 
> You will probably be able to get help here: https://community.platformio.org

Ok i will see tomorrow thanks

Anugaradhawa, 2021-12-15:
[image]

platformio not working

cgreening, 2021-12-15:
Have you posted this problem on the platformio community to get support from them?

I suspect it is because you have spaces in your path (your windows username has spaces) which will break a lot of tools.

Check the platformio community and see if they have a suggestion for you.

Anugaradhawa, 2021-12-15:
> I suspect it is because you have spaces in your path (your windows username has spaces) which will break a lot of tool

I will try to remove space between my username 👍🏻😊

Anugaradhawa, 2021-12-16:
[image]

Thanks, it's successfully installed but there is another error named "not found"

[image]

Anugaradhawa, 2021-12-17:
thanks, you are a genius, it's working pretty now 
[image]
[image]
I found the problem, the problem is not installed platformio correctly and I did not add python to the path when installed

closed 2021-12-20 by Anugaradhawa