atomic14/diy-alexa #35: Code functions

Anugaradhawa, 2021-12-14:
[image]

I my previous issue you tell me to understand the code , therefore i open this issue for get some idea of functions in code . Ok tell me why are you use tf micro folder and src folder in this code??

Anugaradhawa, 2021-12-15:
[image]

why are you use this command "xxd -i converted_model.tflite > model_data.cc"

cgreening, 2021-12-15:
Have you read the text in the screenshot you posted? It explains what this command is used for :)

Can you tell me which part of the text is unclear?

You need to convert the trained model into C code so that it can be embedded in the firmware and run on the ESP32.

I would like to think carefully on if this project is suitable for you. It is an advanced project that requires you to know python, machine learning and embedded development. I cannot teach you all of these things.

I would recommend that you just try and get the existing firmware working before you try and train a new wakeword.

Anugaradhawa, 2021-12-15:
> I would recommend that you just try and get the existing firmware working before you try and train a new wake word

Ok thanks for your comment, I am 14 years old yet, therefore  i think i can not understand the code i have also study therefore mom limited the internet therefore i can't watch the long tutorials but leisure time i try projects using ESP 32.

cgreening, 2021-12-15:
No problem - I will help where I can. But definitely getting platformio up and running and trying the existing firmware would be a good start.

Anugaradhawa, 2021-12-15:
> No problem - I will help where I can. But definitely getting platformio up and running and trying the existing firmware would be a good start.

Where  i start to make a firmware , i haven't a idea.

Anugaradhawa, 2021-12-17:
[image]

I use an I2S microphone then am I need to remove analog microphones ADC code from the firmware folder??

Anugaradhawa, 2021-12-20:
[image]

I use an I2S microphone then am I need to remove analog microphones ADC code from the firmware folder?? What programming language you used in this audio input file??

cgreening, 2021-12-20:
This file contains all the configuration you should need - https://github.com/atomic14/diy-alexa/blob/master/firmware/src/config.h

You should not delete any code.

closed 2021-12-20 by cgreening

Anugaradhawa, 2021-12-21:
> This file contains all the configuration you should need - https://github.com/atomic14/diy-alexa/blob/master/firmware/src/config.h
> 
> You should not delete any code.

Ok i understood