atomic14/diy-alexa #37: Input function

Anugaradhawa, 2021-12-20:
[image]

I use an I2S microphone then am I need to remove analog microphones ADC code from the firmware folder?? What programming language you used in this audio input file??

Anugaradhawa, 2021-12-20:
What programming language you used in this audio input folder ??

cgreening, 2021-12-20:
Everything is controlled from this file: https://github.com/atomic14/diy-alexa/blob/master/firmware/src/config.h

You don't need to delete anything. Edit the config.h file for your hardware setup and then just use platformio to program your ESP32 board.

closed 2021-12-20 by cgreening

Anugaradhawa, 2021-12-21:
> Everything is controlled from this file: https://github.com/atomic14/diy-alexa/blob/master/firmware/src/config.h
> 
> You don't need to delete anything. Edit the config.h file for your hardware setup and then just use platformio to program your ESP32 board.

Thanks