atomic14/diy-alexa #38: Code functions

Anugaradhawa, 2021-12-21:
What programming language is you used  for made the in input folder code?? Python or C++

cgreening, 2021-12-21:
Do you mean the code that runs on the ESP32? That is C++.

Anugaradhawa, 2021-12-22:
> Do you mean the code that runs on the ESP32? That is C++.

Ok thanks I thought it was python

Anugaradhawa, 2021-12-22:
[image]
 

Why we use a ring buffer?? For what purpose??

Anugaradhawa, 2021-12-24:
[image]

What purpose we need to used a ring buffer??

cgreening, 2021-12-24:
We are keeping the latest 1 second worth of audio data samples. The ring buffer is an efficient way to store these sample without having to move them around in memory when more samples arrive.

Anugaradhawa, 2021-12-25:
> We are keeping the latest 1 second worth of audio data samples. The ring buffer is an efficient way to store these sample without having to move them around in memory when more samples arrive.

Thanks, I understand. merry Christmas and Happy new year

closed 2022-01-01 by Anugaradhawa