atomic14/diy-alexa #42: HTTP response status is 200 but content is empty

yanxuceo, 2021-12-31:
Hi Chris, thank you for much for this open source project. You might have heard my voice if you checked your wit.ai app for this project. 

I set up my hardware and flashed this code, then it works well for "Marvin" wake-up and following "tell me a joke" "what's the life" command. 

**The issue is**:
But when I replace the URL and access_key with my settings(which works well with curl and my local recorded .wav samples), I don't get the expected JSON content, I added this debugging line in getResults(), but the returned entities, intents and text are all empty.

`
if (status == 200)
{
        char temp[1024];
        int read_cnt = m_wifi_client->readBytes(temp, 1024);
         Serial.printf("Http str  is: %s\n", temp);
}
`

Do you possibly have any clue? I am really confused since the only difference is the replacement with my wit.ai settings. And in my app, I did receive the recorded and uploaded voice sample. But it seems that the HTTP response got something wrong. 

Thank you so much for any guidance for debugging.

Best regards,
Xu

yanxuceo, 2022-01-01:
I see, it's because now(with an update in October 2021) the /speech endpoint returns partial transcripts, which is mentioned in this issue:
https://github.com/wit-ai/wit/issues/2206

cgreening, 2022-01-01:
That's annoying. I guess we can apply the fix mentioned in that issue. I wonder why they changed it?

yanxuceo, 2022-01-01:
Hi Chris, thanks for the quick reply. I am new to the web stuff, but I will try to fix it. My new confusion is that, why this update has no effect on your wit.ai app(everything works well with your acess_key)?  Does this API update only apply to newly created app?

Or actually your code already handled it correctly, I might make something wrong with my configuration...

I really appreciate your help.

Best regards,
Xu

cgreening, 2022-01-01:
Hi Xu,

I need to plug mine back together - it got taken apart for a new project and I need to re-assemble it.

I'll have a look this weekend as it sounds like a simple fix.

Best
Chris

yanxuceo, 2022-01-01:
Hi Chris, I just learned and fixed something. This is what I get on Terminal when I run:

`
$ curl -XPOST 'https://api.wit.ai/speech?v=20211231' -i -L -H "Authorization: Bearer my_access_key" -H "Content-Type: audio/wav" --data-binary "@apple.wav"
`

**JSON return**:
HTTP/1.1 100 Continue
Date: Sat, 01 Jan 2022 12:29:54 GMT

HTTP/1.1 200 OK
Content-Type: application/json
Date: Sat, 01 Jan 2022 12:29:54 GMT
Transfer-Encoding: chunked
Connection: keep-alive

{
  "text": "Hey"
}
{
  "text": "Hey Facebook"
}
{
  "text": "This is Apple."
}
{
  "entities": {
    "fruit_type:fruit_type": [
      {
        //ignored
      }
    ]
  },
  "intents": [
    {
      //ignored
    }
  ],
  "text": "This is Apple.",
  "traits": {}

For this chunked return, there is no "Content-Length" in the header part. Then in the code, I added these lines after if(status == 200), which I learned from Arduino community:

        String chunk = "";
        int limit = 1;
        String response="";

        Serial.println(response);
        do {
            if (m_wifi_client->connected()) {
            m_wifi_client->setTimeout(4000);
            chunk = m_wifi_client->readStringUntil('\n');
            response += chunk;
            Serial.println(chunk);
        }
        } while (chunk.length() > 0 && ++limit < 1000);
    
Then, I see exact the same JSON return as it is with above curl command.