atomic14/diy-alexa #47: Can you reuse pin for audio input and output ?

hyansuper, 2022-04-10:
Hi, first I want to thank you for the series of tutorial videos on youtube.
If I configure input as left channel, and output as right channel, I wonder if I can share clock/word wires for both audio input and output? something like below
```
static const i2s_pin_config_t pin_config = {
    .bck_io_num = 4,
    .ws_io_num = 5,
    .data_out_num = 18,
    .data_in_num = 17,
};
```

Or, config pins separately, but use same pins for bck_io_num and ws_io_num :
```
// input
static const i2s_pin_config_t pin_config_0 = {
    .bck_io_num = 4,
    .ws_io_num = 5,
    .data_out_num = 18,
    .data_in_num = I2S_PIN_NO_CHANGE
};
i2s_set_pin(i2s_num_0, &pin_config_0);

// output
static const i2s_pin_config_t pin_config_1 = {
    .bck_io_num = 4,
    .ws_io_num = 5,
    .data_out_num = I2S_PIN_NO_CHANGE,
    .data_in_num = 17
};
i2s_set_pin(i2s_num_1, &pin_config_1);
```

Anugaradhawa, 2022-05-06:
hey, can we use an ADC microphone instead of an i2s microphone?

cgreening, 2022-05-07:
@hyansuper I think this is possible, but would require some changes to the code.

At the moment the I2S input and output are always running you would probably need to stop and start them so there wouldn't be any conflict.

This might not be too difficult to do as we know when we have detected a work so we can turn off the input and turn on the output. And we know when we've finished processing a command so we could turn off the input and turn on the input then,

ValeraDanger, 2022-11-01:
> hey, can we use an ADC microphone instead of an i2s microphone?

Yeap, check this [video](https://www.youtube.com/watch?v=pPh3_ciEmzs&t=331s)