atomic14/diy-alexa #51: DIY Alexa is not response

Anugaradhawa, 2022-05-01:
[image]

I correctly upload the codes but it's doesn't show any response when I am saying  "Marvin" please help me I am waiting for your response I can't understand what is  the reason for this please help me

cgreening, 2022-05-07:
The first thing to do is confirm that your microphone is working correclty.

This is a very simple Arduino sketch that will let you test this. Just update the code to match the pins you have connected the microphone to.

https://github.com/atomic14/esp32-i2s-mic-test/tree/main/i2s_mic_test

Anugaradhawa, 2022-05-07:
> The first thing to do is confirm that your microphone is working correclty.
> 
> This is a very simple Arduino sketch that will let you test this. Just update the code to match the pins you have connected the microphone to.
> 
> https://github.com/atomic14/esp32-i2s-mic-test/tree/main/i2s_mic_test 

Yes i find the problem microphone is not working thanks your reply friend.i ordered another microphone i wait for come it

Anugaradhawa, 2022-05-07:
I have a issue . how you convert tf lite model into c code??

Anugaradhawa, 2022-05-07:
[image]

I have also another question . why you use "m_ " prefix in c code what it's purpose??

cgreening, 2022-05-12:
It's a way to indicate to the reader that the variable is a member of the class.

https://stackoverflow.com/questions/1228161/why-use-prefixes-on-member-variables-in-c-classes/1229360#1229360

closed 2022-05-12 by cgreening

Anugaradhawa, 2022-05-12:
> It's a way to indicate to the reader that the variable is a member of the class.
> 
> https://stackoverflow.com/questions/1228161/why-use-prefixes-on-member-variables-in-c-classes/1229360#1229360

Thanks friend