atomic14/diy-alexa #57: detect wakeword process not working

Anugaradhawa, 2022-10-03:
I connect all pins correctly to the esp32 from the microphone and I said "Marvin, Marvin, Marvin" but it does not respond to it.please suggest a method to work correctly

Domnide, 2023-07-31:
Same I've followed every step in the process shown in including training my own model and then Making my own wit.ai app and after all of this the wake word is not bein detected. I've changed 3 mics and even tried your own code also as it is if not the wit.ai api then atleast it should detect the wake word neither in my code nor in yours and not even after changing mics. 
Please help me with this. I've putted in a lot of hardwork and time to get to here. I'll be very obliged if you guide me through this. Thank you.

Domnide, 2023-07-31:
> Same I've followed every step in the process shown in including training my own model and then Making my own wit.ai app and after all of this the wake word is not bein detected. I've changed 3 mics [...]

PS: I've kept the speaker definitions but removed the speaker and connections but still no progress.