atomic14/diy-alexa #60: INMP441 not working

icedifferent, 2023-03-07:
hello, Thank you for your open source.

board : esp32 wroom
microphone: INMP441 

I complie this code to my  esp32, but i find the predict output alway is 0.21 when i say Marvin.
[image]
[image]
so  i try to test my microphone using your i2s_mic_test code , i think this microphone is working.
[image]
https://github.com/atomic14/esp32-i2s-mic-test/blob/main/i2s_mic_test/i2s_mic_test.ino

so how can i find where is go wrong?

sarasqz, 2023-04-23:
Hello, I've been facing the exact same issue. Did you happen to find a solution for this?

Domnide, 2023-07-31:
I am having the same issue can anyone please help me?

sarasqz, 2023-08-04:
Try changing the channel the mic is using in the config.h file. so it should be "#define I2S_MIC_CHANNEL I2S_CHANNEL_FMT_ONLY_RIGHT"