atomic14/diy-alexa #61: ADC-related compile errors when shifting to ESP32S3

hamishcunningham, 2023-05-02:
hi Chris! Some of my students are doing Marvin projects again, and I realise the ADC has changed on the S3 :(
Have you seen this in any of your projects? I've been trying to fix compile errors like `lib/audio_input/ADCSampler.cpp:16:5: error: 'i2s_adc_enable' was not declared in this scope` but not having much luck and perhaps would need a more thorough understanding of the ADC peripheral library in IDF...
Best, Hamish

cgreening, 2023-05-02:
They deprecated a bunch of the old APIs in the latest idf. You can include
the legacy headers I think.

hamishcunningham, 2023-05-03:
According to the docs the old API remains in the same place, with new API in new headers. I think the issue is that the S3 also doesn't support some of the old API so they've removed it. Apparently the S3 doesn't support I2S on ADC2, so functions calls `i2s_set_adc_mode(m_adcUnit, m_adcChannel);` aren't valid any more, at least in the old API.

cgreening, 2023-05-03:
Ah, you are correct -
https://github.com/espressif/esp-idf/issues/8536

Thanks annoying.

sanyi, 2024-02-24:
If ADC is not used, just delete `lib/audio_input/ADCSampler.cpp` and `lib/audio_input/ADCSampler.h` and reference to the header in main. Comment out `REG_SET_BIT(I2S_TIMING_REG(getI2SPort()), BIT(9));
        REG_SET_BIT(I2S_CONF_REG(getI2SPort()), I2S_RX_MSB_SHIFT);` then delete `I2S_MODE_ADC_BUILT_IN` in main. Now your code is messed for good, but works :) However my audio feedback is not right on S3, there is sound that resembles the sound file but is more of a noise. Still trying to figure out if is software or i did something silly with hardware or choosing of the pins.