atomic14/diy-alexa #66: Program is resetting continuously

k-shoeb, 2024-03-14:
I have changed the program a little bit to control a car. And when i am running the code, it reboots continuously.
This is the output from the serial monitor. It is getting Reset after executing the _Starting i2s_ part of code. 
Can anyone tell me why this is happening.

`ets Jul 29 2019 12:21:46

rst:0x8 (TG1WDT_SYS_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:1184
load:0x40078000,len:13232
load:0x40080400,len:3028
entry 0x400805e4
Starting up
Total heap: 284948
Free heap: 214484
CPU frequency: 240
Loading model
12 bytes lost due to alignment. To avoid this loss, please make sure the tensor_arena is 16 bytes aligned.
Used bytes 22604

**_Created Neural Net
m_pooled_energy_size=43
Created audio processor
Starting i2s_**
ets Jul 29 2019 12:21:46
`

ArifulH8, 2024-04-28:
Any update @k-shoeb ?

k-shoeb, 2024-04-28:
> Any update @k-shoeb ?

Issue is solved

The problem with the code was that I had added code for an Ultrasonic Sensor out of a conditional loop. 
Now it's working :)